v-zhuravlev/libzbxmodbus #9: zabbix 3.2

KanstantinM, 2016-09-28:
Dear Vitaly,

Thanks for useful module. Unfortunately it doesn't work with zabbix 3.2. 
Do you have any plans to add support of 3.2 version? 

Thanks.

v-zhuravlev, 2016-09-30:
Yes, stay tuned

v-zhuravlev, 2016-09-30:
Please test and tell me how it goes:
https://github.com/v-zhuravlev/libzbxmodbus/commit/f20e858ac1f0411f83866e5c2e921d2d457583a0

KanstantinM, 2016-10-04:
unfortunately, I was unable to execute autogen.sh. Here is the error: 

automake: warnings are treated as errors
/usr/share/automake-1.13/am/ltlibrary.am: warning: 'libzbxmodbus.la': linking libtool libraries using a non-POSIX
/usr/share/automake-1.13/am/ltlibrary.am: archiver requires 'AM_PROG_AR' in 'configure.ac'
src/Makefile.am:1:   while processing Libtool library 'libzbxmodbus.la'
src/Makefile.am: installing './depcomp'
autoreconf: automake failed with exit status: 1

Thanks.

v-zhuravlev, 2016-10-04:
I think it has nothing to do with changes I made. Please check that you have all build tools

KanstantinM, 2016-10-04:
yes, everything is in place. 

[root@epbygrow0081t9 /]# rpm -qa | grep -e "autoconf\|automake\|libtool\|pkg"
autoconf-2.69-11.el7.noarch
pkgconfig-0.27.1-4.el7.x86_64
libtool-ltdl-2.4.2-21.el7_2.x86_64
automake-1.13.4-3.el7.noarch
libtool-2.4.2-21.el7_2.x86_64

v-zhuravlev, 2016-10-04:
ok , try again: https://github.com/v-zhuravlev/libzbxmodbus/commit/46f36db568cfa9b844f8af71a6f25cdf84a8e353

KanstantinM, 2016-10-05:
it seems to be ok now. Thank you so much. )

closed 2016-10-05 by KanstantinM