v-zhuravlev/libzbxmodbus #11: Timeouts

tonyskalski, 2016-10-28:
Hi and thanks for a great module!

I read the other issue (now closed) related to timeouts, but I still have a problem. I am reading some values through a modbus gateway running on PowerLogic electric meter. If I query the meter itself, I get results. If I query a 'gateway' device (connected to the meter via RS-485) it times out.

I am using libmodbus 3.1.2 as recommended in the other timeout issue. I have timed the responses using tester.exe from Schneider Electric and they take ~2000 ms typically. I can also reproduce the issue with modpoll.exe - with the default timeout (1 sec) I usually receive a connect time-out error. If I increase the timeout to 2-3 seconds I get a value.

I have looked through the libmodbus code to see if there was an easy or obvious way to adjust, but I am not a programmer. Is there a way to set a timeout? Even a global option would help. Thanks!

v-zhuravlev, 2016-10-28:
What error do you have in Zabbix (can check by clicking Red cross in 'Latest data' or in zabbix-*.log)? What does it say? 

So it seems that the default libmodbus timeout is around 1 second or below it, right? 

What Timeout do you have in Zabbix config? 3 seconds as by default?

v-zhuravlev, 2016-10-28:
I guess that default timeout is 0.5s:
https://github.com/stephane/libmodbus/blob/master/src/modbus-private.h#L38

tonyskalski, 2016-10-29:
2472:20161028:150355.123 item "ION
7550:modbus_read_registers[xxx.xxx.xxx.xxx, 5, 3, 3]" became not supported:
Connection timed out

I get the same thing with the default, and also when 'Timeout=10'.

When I get back to the office Monday I will try adjusting
https://github.com/stephane/libmodbus/blob/master/src/modbus-private.h#L38,
re-make libmodbus and re-test. For my environment, having this set globally
will not be a problem.

ajs

v-zhuravlev, 2016-10-29:
Adjust here and uncomment:
https://github.com/v-zhuravlev/libzbxmodbus/blob/master/src/modbus.c#L290
Will have the same effect

v-zhuravlev, 2016-10-31:
I added a commit with module version that uses Timeout from zabbix_agent file. 
Would be great if you could give it a test
https://github.com/v-zhuravlev/libzbxmodbus/commit/09d619d1e40fa26b5df28e0d03e2a9c4c5a52755

tonyskalski, 2016-10-31:
> 09d619d

Tested successfully. Thanks!

This morning I tested by manually setting the timeout here
https://github.com/v-zhuravlev/libzbxmodbus/blob/master/src/modbus.c#L290,
but this is much better.

commit b2b68664 (2016-10-31): Merge pull request #12 from v-zhuravlev/timeout_test

fixes #11 
version 0.5
closed 2016-10-31 by v-zhuravlev