v-zhuravlev/libzbxmodbus #15: Не могу заставить нормально работать.

brutevinch, 2017-06-02:
Происходит рандомно вот такая ерунда:

[image]

v-zhuravlev, 2017-06-02:
Hi,
- Do you use Modbus TCP device or Modbus enc? 
- What version of zabbix and what version of module?
- What Timeout is used in configuration file of Zabbix server? Try increasing it to 10s or more.

brutevinch, 2017-06-02:
1) Modbus TCP
2) Zabbix 3.0
3) current Timeout is set to 10

v-zhuravlev, 2017-06-02:
- what is the device/gateway that you poll?
- what version of libzbxmodule ?

brutevinch, 2017-06-02:
1. selfmade modbus tcp -> modbus rtu gateway on esp8266 (OpenHab polling fine without any problems)
2. last from master branch

brutevinch, 2017-06-02:
Here is openhanded config (maybe it can help you) 

poll=30000
writemultipleregisters=true

tcp.sdm230_Voltage.connection=10.0.1.23:502:0
tcp.sdm230_Voltage.type=input
tcp.sdm230_Voltage.valuetype=float32
tcp.sdm230_Voltage.length=2
tcp.sdm230_Voltage.start=0
tcp.sdm230_Voltage.id=1

tcp.sdm230_Current.connection=10.0.1.23:502:0
tcp.sdm230_Current.type=input
tcp.sdm230_Current.valuetype=float32
tcp.sdm230_Current.length=2
tcp.sdm230_Current.start=6
tcp.sdm230_Current.id=1

tcp.sdm230_Power.connection=10.0.1.23:502:0
tcp.sdm230_Power.type=input
tcp.sdm230_Power.valuetype=float32
tcp.sdm230_Power.length=2
tcp.sdm230_Power.start=12
tcp.sdm230_Power.id=1

v-zhuravlev, 2017-06-02:
It might be related to resource locking (there is no for TCP).

Can you try the following branch (with locking enabled for TCP as well)?

`git clone -b TCP-always-lock --recursive https://github.com/v-zhuravlev/libzbxmodbus.git`
Run autogen.sh
`./configure --prefix=/etc/zabbix --enable-zabbix-3`
`make`
`make install`

brutevinch, 2017-06-02:
Same thing =(

v-zhuravlev, 2017-06-02:
1. Ok, to 100% make sure it has nothing to do with resource sharing:
In your Zabbix server (if it is possible to do so) please leave only single poller: 
StartPollers=1 and restart.

2. can you also run `ipcs -s` while your Zabbix server with module loaded is runnning? and show me the output of it.

brutevinch, 2017-06-02:
1. done
2. 
`/usr/src/libzbxmodbus(TCP-always-lock*) # ipcs -s                                                                                                 root@Synology

------ Semaphore Arrays --------
key        semid      owner      perms      nsems     
0x0052e2c1 0          55         600        17        
0x0052e2c2 32769      55         600        17        
0x0052e2c3 65538      55         600        17        
0x20161122 98307      root       666        1         
0x7a028b43 557060     zabbix     600        13        
0x7a021f65 589829     zabbix     600        1         
0x7a020152 196614     zabbix     600        13        
0x00000000 262151     1023       600        1  `

v-zhuravlev, 2017-06-02:
Strange, accodring to this it looks like that you might be using old version of the module anyway.
In the new version you must see zabbix semaphore with nsems = 128 and not 1.

Try deleting your old modules from (/usr/local/lib i suppose)
and in zabbix server config point to the new location: 

LoadModulePath = /etc/zabbix/lib

Restart server. Check that `ipcs -s` shows you the semaphore with nsems = 128.
If not going to work, which linux distro do you use?

brutevinch, 2017-06-02:
Here is with new module:
0x7a021f65 655365     zabbix     600        128       

in zabbix 2 of 3 data sources are "Connection timed out"

v-zhuravlev, 2017-06-02:
even with single poller?

brutevinch, 2017-06-02:
Ну если 1 поллер включается добавлением в конфиг сервера то да. Вот кусок конфига сервера:
LoadModulePath = /etc/zabbix/lib
LoadModule = libzbxmodbus.so
Timeout = 10

StartPollers=1

v-zhuravlev, 2017-06-02:
uff....weird.

Ok, lets try it again: 

`git clone -b debug_enabled --recursive https://github.com/v-zhuravlev/libzbxmodbus.git`
Run autogen.sh
`./configure --prefix=/etc/zabbix --enable-zabbix-3`
`make`
`make install`

Then in configuration file:
```
LogType=console
DebugLevel=1
```
In zabbix frontend: Disable all modbus items and leave only one.

Stop the zabbix-server
rerun it with in the foreground:
`zabbix_server -f`

You should now see all the modbus debug messages from the libmodbus lib. Please attach it here. They should contain a part where it fails also.

brutevinch, 2017-06-02:
~ # telnet 10.0.1.23 502                                                                                                                          
Trying 10.0.1.23...
Connected to 10.0.1.23.
Escape character is '^]'.

Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][06][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><3E><27><37><A8>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][0C][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><41><FA><04><EE>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][00][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><43><6A><ED><31>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][06][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><3E><2B><F3><48>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][0C][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><41><F8><64><7A>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][00][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><43><6C><15><97>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][06][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><3E><31><09><4E>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][0C][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><41><F7><19><43>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][0C][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><41><F6><B6><01>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][00][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><43><6B><8B><74>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][06][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><3E><2F><97><4A>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][0C][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><41><F8><48><6D>

v-zhuravlev, 2017-06-02:
You don't have any errors/timeouts with single item?

brutevinch, 2017-06-02:
testing now. tell you later

v-zhuravlev, 2017-06-02:
ok. If there are no issues with single item -> try reenabling them

brutevinch, 2017-06-02:
with enabled 2 items:

Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><43><65><5A><85>
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][06][00][02]
Waiting for a confirmation...
<25><E3><00><00><00><07><01><04><04><43><65><C7><38>
Invalid transaction ID received 0x25E3 (not 0x1)
Connecting to 10.0.1.23:502
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][00][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><43><63><3F><22>

brutevinch, 2017-06-02:
here is with 3 items:

v-zhuravlev, 2017-06-02:
It looks like that there is still more than one poller. Maybe  do you have another StartPollers somewhere in the config file with not 1?
try: in your debug_enabled repo:
`git pull`
`make install`
new logging of zabbix_server -f now should provide `pid` of the process.

brutevinch, 2017-06-02:
here is full config:

LogFile=/var/log/zabbix/zabbix_server.log
LogFileSize=0
LogType=console
DebugLevel=1
PidFile=/var/run/zabbix/zabbix_server.pid
DBName=zabbix
DBUser=zabbix
DBPassword=XXXX
AlertScriptsPath=/usr/lib/zabbix/alertscripts
ExternalScripts=/usr/lib/zabbix/externalscripts
FpingLocation=/usr/bin/fping
Fping6Location=/usr/bin/fping6
LogSlowQueries=3000

LoadModulePath = /etc/zabbix/lib
LoadModule = libzbxmodbus.so
Timeout = 10

StartPollers=1
AllowRoot = 1

Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][06][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><3E><2F><C6><F6>
pid:2247 hash:118 : 10.0.1.23 Unlocking 
pid:2247 hash:118 : 10.0.1.23 Locking 
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][0C][00][02]
Waiting for a confirmation...
ERROR Connection timed out: select
pid:2247 hash:118 : 10.0.1.23 Unlocking 
pid:2247 hash:118 : 10.0.1.23 Locking 
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][00][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><43><5F><A8><25>

v-zhuravlev, 2017-06-02:
how did you implement Modbus TCP? It this your own code?

brutevinch, 2017-06-02:
2247 hash:118 : 10.0.1.23 Unlocking 
pid:2247 hash:118 : 10.0.1.23 Locking 
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][06][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><3E><32><91><DE>
pid:2247 hash:118 : 10.0.1.23 Unlocking 
pid:2247 hash:118 : 10.0.1.23 Locking 
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][00][00][02]
Waiting for a confirmation...
<28><55><00><00><00><07><01><04><04><43><61><FE><51>
Invalid transaction ID received 0x2855 (not 0x1)
pid:2247 hash:118 : 10.0.1.23 Unlocking 
pid:2247 hash:118 : 10.0.1.23 Locking 
Connecting to 10.0.1.23:502
pid:2247 hash:118 : 10.0.1.23 Unlocking 
pid:2247 hash:118 : 10.0.1.23 Locking 
Connecting to 10.0.1.23:502
pid:2247 hash:118 : 10.0.1.23 Unlocking 
pid:2247 hash:118 : 10.0.1.23 Locking 
Connecting to 10.0.1.23:502
pid:2247 hash:118 : 10.0.1.23 Unlocking 
pid:2247 hash:118 : 10.0.1.23 Locking 
Connecting to 10.0.1.23:502
[00][01][00][00][00][06][01][04][00][06][00][02]
Waiting for a confirmation...
<00><01><00><00><00><07><01><04><04><3E><30><ED><CE>
pid:2247 hash:118 : 10.0.1.23 Unlocking

brutevinch, 2017-06-02:
Here is gateway code 

[modbus_gate_v2.lua.zip](https://github.com/v-zhuravlev/libzbxmodbus/files/1047885/modbus_gate_v2.lua.zip)

v-zhuravlev, 2017-06-02:
>selfmade modbus tcp -> modbus rtu gateway on esp8266 (OpenHab polling fine without any problems)

Could it be that OpenHab actually also sometimes receive invalid data/timeout? Just it doesn't show you that?

brutevinch, 2017-06-02:
Saw openhab traces. There is no errors and warnings and all data received every 30 sec without any errors too. =(

brutevinch, 2017-06-02:
Every 30 sec.

[image]

v-zhuravlev, 2017-06-02:
So let's sum ip what we have:

- no problem with single poller and single item enabled
- problem with single poller and multiple items
- problem with multiple pollers and multiple items

v-zhuravlev, 2017-06-02:
try with 0 byte timeout:

in your debug_enabled repo:
`git pull`
`make install`

brutevinch, 2017-06-03:
Hi, now I see some errors in openhab, please give some time to test it.

brutevinch, 2017-06-06:
Sorry it was unstable buggy code.

closed 2017-06-06 by brutevinch

v-zhuravlev, 2017-06-06:
@brutevinch , Did you fix it? Is it working with libzbxmodbus now? With(debug branch) or without(current main branch) TCP locking?

brutevinch, 2017-06-09:
Rewrite gateway code to mqtt.