v-zhuravlev/libzbxmodbus #18: INT64 and FLOAT64 is Supported?

peruchi, 2017-07-27:
Good afternoon, I'm using an equipment that has a register 64bits, INT64 and FLOAT64, and I can not capture the data, is it not supported?

v-zhuravlev, 2017-07-28:
No, only 16 and 32 bits.

peruchi, 2017-07-28:
I need to monitor the pm5340 equipment. Would you have a forecast for this implementation?

v-zhuravlev, 2017-07-29:
Do you have Modbus spec for pm5340? With registers adresses?

peruchi, 2017-07-31:
Yes, follow the [link](http://www2.schneider-electric.com/resources/sites/SCHNEIDER_ELECTRIC/content/live/FAQS/234000/FA234017/en_US/PM5100_PM5300_ModbusRegisterList.xls).

I'm trying to monitor the items:

Active Energy Delivered (Into Load) |   |   | 3204 | Wh | 4 | INT64
Active Energy Received (Out of Load) |   |   | 3208 | Wh | 4 | INT64
Active Energy Delivered + Received |   |   | 3212 | Wh | 4 | INT64
Active Energy Delivered – Received |   |   | 3216 | Wh | 4 | INT64
Reactive Energy Delivered |   |   | 3220 | VARh | 4 | INT64
Reactive Energy Received |   |   | 3224 | VARh | 4 | INT64
Reactive Energy Delivered + Received |   |   | 3228 | VARh | 4 | INT64
Reactive Energy Delivered – Received |   |   | 3232 | VARh | 4 | INT64
Apparent Energy Delivered |   |   | 3236 | VAh | 4 | INT64
Apparent Energy Received |   |   | 3240 | VAh | 4 | INT64
Apparent Energy Delivered + Received |   |   | 3244 | VAh | 4 | INT64
Apparent Energy Delivered – Received |   |   | 3248 | VAh | 4 | INT64

And this Item (Power Factor Total 3084 4Q FP PF), I had to make two minus the value received.

v-zhuravlev, 2017-07-31:
You mentioned Float64?

peruchi, 2017-07-31:
I mentioned Float64, however there is not a register that needs it at this point, but INT64 would be what I need. Would implementation be possible?

peruchi, 2017-08-07:
Hi, this would implementation be possible?

v-zhuravlev, 2017-08-07:
Yes, its possible, but can't do it right now. Will try to do it by the end of the week.

peruchi, 2017-08-18:
Sorry, but would you have a prediction of when you will be able to deploy?

v-zhuravlev, 2017-08-18:
Sorry! Really busy at the moment. I'll try to address your request till the end of the next week

peruchi, 2017-08-18:
tks.

v-zhuravlev, 2017-10-04:
please test this branch: https://github.com/v-zhuravlev/libzbxmodbus/tree/int64
use datatype - `S` for INT64 and `d` for FLOAT64. In Zabbix use Numeric(float). Or, if you are sure that this register never goes below zero - use Numeric

Also, please use modpoll utility(http://www.modbusdriver.com/modpoll.html
) and poll this first INT64 you mentioned from power meter with keys similar to this:
modpoll.exe -t 3:hex -a <slave address> -r 3204 -c 4 -1 127.0.0.1
see help -h for additional help on this utility

```
modpoll.exe -t 3:hex -a 4 -r 8 -c 4 -1 127.0.0.1
modpoll 3.4 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2013 proconX Pty Ltd
Visit http://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 4, start reference = 8 (PDU), count = 4
Communication.........: 127.0.0.1, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register (hex), input register table

-- Polling slave...
[8]: 0xABCD
[9]: 0xFE12
[10]: 0x0908
[11]: 0x43F0
```
With hexes provided, please tell me what is the expected INT64 value here.

Thanks!

peruchi, 2017-10-04:
Good Morning,

I tested, but not getting values.

In Zabbix: modbus_read[192.168.0.125,255,3208,3,s,1,0]

In Modpoll: 

./modpoll -t 4:hex -a 255 -r 3212 -c 1 -1 192.168.0.125
modpoll 3.4 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2013 proconX Pty Ltd
Visit http://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 255, start reference = 3212, count = 1
Communication.........: 192.168.0.125, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register (hex), output (holding) register table

-- Polling slave...
[3212]: 0x0000

./modpoll -t 4:int -a 255 -r 3212 -c 1 -1 192.168.0.125
modpoll 3.4 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2013 proconX Pty Ltd
Visit http://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 255, start reference = 3212, count = 1
Communication.........: 192.168.0.125, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 32-bit integer, output (holding) register table

-- Polling slave...
[3212]: 0

The correct value is 1.7483

Am I setting it wrong?

v-zhuravlev, 2017-10-04:
in zabbix use capital S.
in modpoll there is no way to poll int64, so please use hex with -c 4:
./modpoll -t 4:hex -a 255 -r 3212 -c 4 -1 192.168.0.125

v-zhuravlev, 2017-10-04:
> The correct value is 1.7483

I thought it is an INT64? without decimal point

peruchi, 2017-10-04:
So in documentation shows

Active Energy Delivered + Received | Y |   | 3212 | Wh | 4 | INT64

But in equipament shows 1.7739

I've already checked it and that's it, but if you get only the whole value already help =

http://www2.schneider-electric.com/resources/sites/SCHNEIDER_ELECTRIC/content/live/FAQS/234000/FA234017/en_US/PM5100_PM5300_ModbusRegisterList.xls

v-zhuravlev, 2017-10-04:
please run modpoll with -c 4 as above

peruchi, 2017-10-04:
./modpoll -t 4:hex -a 255 -r 3212 -c 4 -1 192.168.0.125
modpoll 3.4 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2013 proconX Pty Ltd
Visit http://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 255, start reference = 3212, count = 4
Communication.........: 192.168.0.125, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register (hex), output (holding) register table

-- Polling slave...
[3212]: 0x0000
[3213]: 0x0000
[3214]: 0x001B
[3215]: 0x3BFB

v-zhuravlev, 2017-10-04:
1.7739 - is it Wh or MWh?

peruchi, 2017-10-04:
at this time MWh
This equipment is test

v-zhuravlev, 2017-10-04:
ok try now

peruchi, 2017-10-04:
Now it worked but with the command you passed the error occurred:

modbus_read[{HOST.CONN},{$MODBUS_SLAVE},3212,4,I]" became not supported: Illegal function

and it worked this way:

modbus_read[{HOST.CONN},{$MODBUS_SLAVE},3212,3,S,1,1]

Can be the equipment

peruchi, 2017-10-04:
Thank you for your help

closed 2017-10-04 by peruchi

v-zhuravlev, 2017-10-04:
hm... how about `modbus_read[{HOST.CONN},{$MODBUS_SLAVE},3212,3,I,1,1]`

peruchi, 2017-10-04:
It's work too.

Result: 1.88 MWh

v-zhuravlev, 2017-10-04:
do you have any float64 to test?

peruchi, 2017-10-04:
I have no register with float64, sorry.

v-zhuravlev, 2017-10-05:
released in master branch (v0.7). Please test and report any issues.