v-zhuravlev/libzbxmodbus #19: Zabbix 3.4 support

darrynv, 2017-09-25:
Hi

Please can you add support for Zabbix 3.4 .

Thank you

v-zhuravlev, 2017-09-25:
Does it not work with 3.2 ? Whats is the error?

darrynv, 2017-09-29:
My apologies. It's working.. I made the incorrect assumption. Please close ;) and thank you!

45928:20170929:142338.812 using configuration file: /etc/zabbix/zabbix_proxy.conf
 45928:20170929:142338.813 loaded modules: libzbxmodbus.so
 45928:20170929:142338.837 current database version (mandatory/optional): 03040000/03040000
 45928:20170929:142338.837 required mandatory version: 03040000
 45928:20170929:142340.992 proxy #0 started [main process]
libzbxmodbus: loaded version: 0.6
 45931:20170929:142340.992 proxy #1 started [configuration syncer #1]
 45932:20170929:142340.993 proxy #2 started [heartbeat sender #1]
 45936:20170929:142340.995 proxy #6 started [http poller #1]

closed 2017-09-29 by v-zhuravlev