v-zhuravlev/libzbxmodbus #21: bulk support

v-zhuravlev, 2017-10-05:
I think that module can be used with bulk mode. So, an array of similar Modbus datatypes can be polled at once. Protocol and libmodbus supports that so what we need in libzbxmodbus to do is return an array as JSON so that can be parsed later with Zabbix 3.4 preprocessing.

v-zhuravlev, 2018-09-05:
Here is the thing:
1. Add new zabbix key: `modbus_read_bulk[<connection>,<slave_id>,<first_reg_to_read>-<last_reg_to_read>,<modbus_function>[,<datatype>][,<endiannes>][,<first_reg>]]`
for example: 
`modbus_read_bulk[/dev/ttyS0,32,4-10,3,float]` to read registers(16bit words) starting from 4 to 9.
2. Convert received registers(16 bit words) into an array of datatype requested.
For example, for `modbus_read_bulk[/dev/ttyS0,32,4-10,3,float]`, you need to convert registers 4-9 16 bit words into 32bit floats:
so new array of float becomes: float[3] array
2.1. it should be checked that number of registers requested is even for 32bit datatypes(uint32, int32, float)
and can be divded by 4 for 64bit datatypes: (uint64, double)
if not, ZBX_NOTSUPPORTED must be returned immediately.

3. New converted array must converted to JSON array and returned to Zabbix as a string. so in our example float[3] array becomes:
```
{
 [ 
  float1,
  float2,
  float3
 ]
}
```
4. In zabbix 3,4+, configure master item with datatype STRING
5. Configure dependent items with jsonpath preprocessing to extract floats like so:
`$.[0]`
6....
7 PROFIT

i-ky, 2018-09-05:
Alternative names for `skip` (a register that is requested from a device as a part of continuous block of data, but is not used to return the result):
1. `blank`
2. `gap`
3. `pass`
4. `void`

v-zhuravlev, 2018-09-05:
I think skip is quite good, lets leave it.

brickamax, 2018-09-18:
Hello
I've tested expression:
modbus_read[tcp://x.x.x.x:51113,2,3905,3,3*float+2*skip+uint64]

Zabbix answer: not supported, check datatype provided..

i-ky, 2018-09-18:
> return an array as JSON

@v-zhuravlev , are you sure about JSON? My concern is that JSON Path support in Zabbix is very limited and perhaps XML would give users more flexibility (e.g. to sum or average values using XPath). However, JSON Path is preferable in the most trivial and the most frequently used cases (simply `$.6` vs. `/value[@register=6]`).

v-zhuravlev, 2018-09-19:
I think that all arrays to be returned from libzbxmodbus are very simple and no problem to parse them with Zabbix flavor of JSONPath. Besides, I still think that 'advanced' JSONPath  syntax will be implemented at some point.

closed 2018-09-23 by v-zhuravlev