v-zhuravlev/libzbxmodbus #22: Конфликтует с системной libmodbus

evgkrsk, 2017-10-07:
0.7 при установке собирает libmodbus-3.1.4 (вроде бы патченую) и устанавливает вместо уже установленной в системе версии. В теми же именами и сонеймами библиотеки. Таким образом, нельзя безопасно установить пакет туда, где уже есть ПО, линкующееся с libmodbus.

С внешней libmodbus пакет тоже разучился собираться, насколько я вижу.

v-zhuravlev, 2017-10-07:
это с 0.6. Что посоветуете?

evgkrsk, 2017-10-07:
Лучше всего продвинуть нужные патчи в апстрим библиотеки и линковаться динамически, требуя версию не ниже нужной.

v-zhuravlev, 2017-10-07:
к сожалению я уже делал pull request, но нет реакции https://github.com/stephane/libmodbus/pull/385

i-ky, 2021-05-03:
Думаю, мы можем класть патченный libmodbus туда же, куда и сам модуль и указывать относительный путь при линковке. Согласно [`man ld.so`](https://man7.org/linux/man-pages/man8/ld.so.8.html):
```
       When resolving shared object dependencies, the dynamic linker
       first inspects each dependency string to see if it contains a
       slash (this can occur if a shared object pathname containing
       slashes was specified at link time).  If a slash is found, then
       the dependency string is interpreted as a (relative or absolute)
       pathname, and the shared object is loaded using that pathname.
```