v-zhuravlev/libzbxmodbus #23: cannot load module "libzbxmodbus.so": /usr/local/lib/libzbxmodbus.so: undefined symbol: modbus_new_rtutcp

Netbulae, 2017-10-24:
After compiling the reset github code, we get this error:

15092:20171024:073500.621 Starting Zabbix Agent [*****]. Zabbix 3.2.6 (revision 67849).
 15092:20171024:073500.622 **** Enabled features ****
 15092:20171024:073500.622 IPv6 support:           NO
 15092:20171024:073500.622 TLS support:            NO
 15092:20171024:073500.622 **************************
 15092:20171024:073500.623 using configuration file: /usr/local/etc/zabbix_agentd.conf
 15092:20171024:073500.629 cannot load module "libzbxmodbus.so": /usr/local/lib/libzbxmodbus.so: undefined symbol: modbus_new_rtutcp
 15092:20171024:073500.629 loading modules failed, exiting...

v-zhuravlev, 2017-10-24:
you are probably linked to original `libmodbus`. but you need patched libmodbus with function `modbus_new_rtutcp`, provided with this module as well. If you used `git clone` make sure that you used `--recusive` flat

Netbulae, 2017-10-31:
yes that was it, thanks

closed 2017-10-31 by Netbulae