v-zhuravlev/libzbxmodbus #26: Unsupported item key.

mxj333, 2017-12-28:
Hello:

I created a items :
```
Name: A V Sensor
Type: Simple check
Key: modbus_read[192.168.0.174,2,57,4,f]
Type of information: Numeric(float)
Units: C

```
After saving, the state of the items is
```
	Not supported
```
info:

```
Unsupported item key.
```

Any idea on how to resolve this problem? 

thank you！

mxj333, 2017-12-28:
It was successful when it was just configured, and it appeared in a few days：
```
Unsupported item key.
```

v-zhuravlev, 2018-01-10:
Hi, could please provide more information? Can you successfully continuously poll you device using `modpoll` utility?

i-ky, 2018-07-10:
Module hasn't been loaded by Zabbix server/proxy/agent I guess. It is worth checking configuration file.

closed 2018-07-10 by v-zhuravlev