v-zhuravlev/libzbxmodbus #27: Timeouts with modbus RTU over RS485

Netbulae, 2018-01-10:
Thanks for the great piece of software, but we keep getting timeouts on modbus RTU. After running the zabbix in debug mode it appears only the the first "modbus_read_registers" of each update gets a timout. After this the item becomes "Not supported" for a while.

We played around with the zabbix agent buffer, baudrate, termination and rs485 settings on linux but the data retrieval remains inconsistent. This is a shame because we would like to use it in production :-& Anything we can try? I've included two samples but it is consistent in that it always fails the first request. Timeout is set to 10

  29199:20180110:144201.371 active check "modbus_read_registers[/dev/ttyS1 19200 N 8 1,26,74,4,f,1,0]" **is not supported: Connection timed out**
 29199:20180110:144201.375 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS6},74,4,f,1,0]' value:'**Connection timed out**'
 29199:20180110:144201.378 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:9/65535
 29199:20180110:144201.381 send_buffer() now:1515595321 lastsent:1515595290 now-lastsent:31 BufferSend:60; will not send now
 29199:20180110:144201.386 End of send_buffer():SUCCEED
 29199:20180110:144201.390 buffer: new element 9
 29199:20180110:144201.394 End of process_value():SUCCEED
 29199:20180110:144201.423 for key [modbus_read_registers[/dev/ttyS1 19200 N 8 1,27,34,4,f,1,0]] received value [-0.992818]
 29199:20180110:144201.426 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS7},34,4,f,1,0]' value:'-0.992818'
 29199:20180110:144201.429 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:10/65535
 29199:20180110:144201.433 send_buffer() now:1515595321 lastsent:1515595290 now-lastsent:31 BufferSend:60; will not send now
 29199:20180110:144201.437 End of send_buffer():SUCCEED
 29199:20180110:144201.441 buffer: new element 10
 29199:20180110:144201.444 End of process_value():SUCCEED
 29199:20180110:144201.446 In need_meta_update() key:modbus_read_registers[/dev/ttyS1 19200 N 8 1,27,34,4,f,1,0]
 29199:20180110:144201.449 End of need_meta_update():FAIL
 29199:20180110:144201.480 for key [modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,72,4,f,1,0]] received value [101.209824]
 29199:20180110:144201.482 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS2},72,4,f,1,0]' value:'101.209824'
 29199:20180110:144201.484 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:11/65535
 29199:20180110:144201.486 send_buffer() now:1515595321 lastsent:1515595290 now-lastsent:31 BufferSend:60; will not send now
 29199:20180110:144201.488 End of send_buffer():SUCCEED
 29199:20180110:144201.491 buffer: new element 11
 29199:20180110:144201.493 End of process_value():SUCCEED
 29199:20180110:144201.496 In need_meta_update() key:modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,72,4,f,1,0]
 29199:20180110:144201.498 End of need_meta_update():FAIL
 29199:20180110:144201.521 for key [modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,0,4,f,1,0]] received value [232.727173]
 29199:20180110:144201.523 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS2},0,4,f,1,0]' value:'232.727173'
 29199:20180110:144201.525 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:12/65535
 29199:20180110:144201.527 send_buffer() now:1515595321 lastsent:1515595290 now-lastsent:31 BufferSend:60; will not send now
 29199:20180110:144201.529 End of send_buffer():SUCCEED
 29199:20180110:144201.531 buffer: new element 12
 29199:20180110:144201.534 End of process_value():SUCCEED
 29199:20180110:144201.537 In need_meta_update() key:modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,0,4,f,1,0]
 29199:20180110:144201.539 End of need_meta_update():FAIL
 29199:20180110:144201.566 for key [modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,30,4,f,1,0]] received value [-0.719867]
 29199:20180110:144201.568 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS2},30,4,f,1,0]' value:'-0.719867'
 29199:20180110:144201.570 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:13/65535
 29199:20180110:144201.572 send_buffer() now:1515595321 lastsent:1515595290 now-lastsent:31 BufferSend:60; will not send now
 29199:20180110:144201.574 End of send_buffer():SUCCEED
 29199:20180110:144201.576 buffer: new element 13
 29199:20180110:144201.579 End of process_value():SUCCEED
 29199:20180110:144201.582 In need_meta_update() key:modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,30,4,f,1,0]
 29199:20180110:144201.584 End of need_meta_update():FAIL
 29199:20180110:144201.712 for key [modbus_read_registers[/dev/ttyS1 19200 N 8 1,23,82,4,f,1,0]] received value [6.112636]
 29199:20180110:144201.714 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS3},82,4,f,1,0]' value:'6.112636'
 29199:20180110:144201.716 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:14/65535
 29199:20180110:144201.718 send_buffer() now:1515595321 lastsent:1515595290 now-lastsent:31 BufferSend:60; will not send now
 29199:20180110:144201.720 End of send_buffer():SUCCEED
 29199:20180110:144201.724 buffer: new element 14
 29199:20180110:144201.727 End of process_value():SUCCEED
 29199:20180110:144201.729 In need_meta_update() key:modbus_read_registers[/dev/ttyS1 19200 N 8 1,23,82,4,f,1,0]
 29199:20180110:144201.732 End of need_meta_update():FAIL
 29199:20180110:144201.813 for key [modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,18,4,f,1,0]] received value [224.861649]
 29199:20180110:144201.815 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS2},18,4,f,1,0]' value:'224.861649'
 29199:20180110:144201.817 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:15/65535
 29199:20180110:144201.819 send_buffer() now:1515595321 lastsent:1515595290 now-lastsent:31 BufferSend:60; will not send now
 29199:20180110:144201.821 End of send_buffer():SUCCEED
 29199:20180110:144201.823 buffer: new element 15
 29199:20180110:144201.826 End of process_value():SUCCEED

29195:20180110:144326.636 __zbx_zbx_setproctitle() title:'collector [processing data]'
 29195:20180110:144326.637 In update_cpustats()
 29195:20180110:144326.639 End of update_cpustats()
 29195:20180110:144326.640 __zbx_zbx_setproctitle() title:'collector [idle 1 sec]'
 29199:20180110:144327.224 active check "modbus_read_registers[/dev/ttyS1 19200 N 8 1,23,26,4,f,1,0]" **is not supported: Connection timed out**
 29199:20180110:144327.225 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS3},26,4,f,1,0]' value:'**Connection timed ou**t'
[... 72 lines cut ...]
 29199:20180110:144327.507 End of send_buffer():SUCCEED
 29199:20180110:144327.507 buffer: new element 104
 29199:20180110:144327.508 End of process_value():SUCCEED
 29199:20180110:144327.508 In need_meta_update() key:modbus_read_registers[/dev/ttyS1 19200 N 8 1,23,22,4,f,1,0]
 29199:20180110:144327.508 End of need_meta_update():FAIL
 29199:20180110:144327.538 for key [modbus_read_registers[/dev/ttyS1 19200 N 8 1,21,60,4,f,1,0]] received value [-1662.992065]
 29199:20180110:144327.538 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS1},60,4,f,1,0]' value:'-1662.992065'
 29199:20180110:144327.539 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:105/65535
 29199:20180110:144327.539 send_buffer() now:1515595407 lastsent:1515595353 now-lastsent:54 BufferSend:60; will not send now
 29199:20180110:144327.540 End of send_buffer():SUCCEED
 29199:20180110:144327.540 buffer: new element 105
 29199:20180110:144327.541 End of process_value():SUCCEED
 29199:20180110:144327.541 In need_meta_update() key:modbus_read_registers[/dev/ttyS1 19200 N 8 1,21,60,4,f,1,0]
 29199:20180110:144327.542 End of need_meta_update():FAIL
 29199:20180110:144327.571 for key [modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,74,4,f,1,0]] received value [80.737862]
 29199:20180110:144327.571 In process_value() key:'zabbix-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS2},74,4,f,1,0]' value:'80.737862'
 29199:20180110:144327.572 In send_buffer() host:'xx.xxx.xx.xxx' port:10051 entries:106/65535
 29199:20180110:144327.572 send_buffer() now:1515595407 lastsent:1515595353 now-lastsent:54 BufferSend:60; will not send now
 29199:20180110:144327.573 End of send_buffer():SUCCEED
 29199:20180110:144327.573 buffer: new element 106
 29199:20180110:144327.574 End of process_value():SUCCEED
 29199:20180110:144327.575 In need_meta_update() key:modbus_read_registers[/dev/ttyS1 19200 N 8 1,22,74,4,f,1,0]
 29199:20180110:144327.575 End of need_meta_update():FAIL
 29195:20180110:144327.640 __zbx_zbx_setproctitle() title:'collector [processing data]'
 29195:20180110:144327.641 In update_cpustats()

v-zhuravlev, 2018-01-10:
Hi, thats strange. Do you test this with a single device or with many devices?

v-zhuravlev, 2018-01-10:
`After running the zabbix in debug mode it appears only the the first "modbus_read_registers" of each update` 
What is `update`?

Netbulae, 2018-01-11:
- We have a setup with a single device that's working fine and a couple of other gateways with 7 meters that work poorly. Even when I disable 6 of the 7.
- zabbix server 3.4, agent 3.2.6 and module latest from git
- Zabbix agent(active)
- Just tried modpoll and it gives 1-3 errors and then gets the data

So maybe it's not an issue with the zabbix agent after all. The only difference between the different systems is that the correctly working one has a USB->RS485 connector and the failing ones have a direct RS485 interface (/dev/ttyUSB0 vs /dev/ttyS1).

I'll investigate. 

Is there a version compatible with zabbix 3.4 agent in the works already? We can help if needed.

> What is update?
Each time the agent polls for modbus data. I see the first poll failing and the rest of the items after that are ok. So that's an update (item _update_ interval) for me.

v-zhuravlev, 2018-01-11:
> Is there a version compatible with zabbix 3.4 agent in the works already? We can help if needed.
Latest libzbxmodule should work with Zabbix agent 3.4. Report if not.

v-zhuravlev, 2018-01-11:
BTW, try using Zabbix agent type , not active.

Netbulae, 2018-01-11:
Ok, I'll compile zabbix agent 3.4 on this and see if it works properly.

> BTW, try using Zabbix agent type , not active.

The gateway is behind a firewall so that's why I use active. I can try if I can poke a hole, but as I said it works the same way on the device with the USB>RS485 converter..

Netbulae, 2018-01-11:
Sorry for the confusion but the zabbix agent has a supervisor program that restarted it. That's why I got the same result as the zabbix output :-&

Tried again with zabbix agent disabled properly and I get normal results using modpoll. No timeouts or anything.

I also disabled all the items, when I enable only 1 or 2 of the items, things work smooth. Enabling 10+ than that triggers communication errors.

Maybe it's indeed and active agent issue, I'll try punching a hole in the firewall and set it to normal Zabbix agent type to see if that helps

Netbulae, 2018-01-11:
I tried it with the normal Zabbix agent type and I now get different errors with +- 30 items enabled.

 30944:20180111:170035.089 resuming Zabbix agent checks on host "zabbix-gw2": connection restored
 30938:20180111:170040.615 Zabbix agent item "modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS3},73,4,f,1,1]" on host "zabbix-gw2" failed: first network error, wait for 15 seconds

And with debug:
 30461:20180111:165705.974 End of get_value_agent():TIMEOUT_ERROR
 30461:20180111:165705.974 Item [idiot-gw2:modbus_read_registers[{$DEVICE} {$SERIAL_SET},{$MOD_ADRESS4},72,4,f,1,0]] error: Get value from agent failed: ZBX_TCP_READ() timed out
 30461:20180111:165705.974 End of get_value():TIMEOUT_ERROR
 30461:20180111:165705.974 In zbx_deactivate_item_host() hostid:10254 itemid:28741 type:0

Still most of the items keep switching between not supported and supported :-(

The normal zabbix os items work fine either way..

v-zhuravlev, 2018-01-25:
- Could you try to increase number of pollers in the Zabbix agent?
- How often do you poll those 30 items?

Netbulae, 2018-01-30:
- You mean StartAgents? I've tried many different settings, currently I have it set to 15.
  Server side, I now have StartPollers=30, but I get a lot of "failed: first network error, wait for 15   seconds"
   Should it be higher?
- We were aiming for every 15 seconds, but increased it to 30s because of issues.

By the way I see this error in the agent log with debug, probably unrelated.

cannot find "zbx_module_history_write_cbs()" function in module "libzbxmodbus.so": /usr/local/lib/libzbxmodbus.so: undefined symbol: zbx_module_history_write_cbs

Netbulae, 2018-01-30:
When I try the local polling and disable the agent checks, I get consistent sub-second responses.
time zabbix_get -s localhost -kmodbus_read["/dev/ttyS1 19200 N 8 1,22,32,4,f,1,0"]
0.989461

real	0m0.075s
user	0m0.008s
sys	0m0.028s

time zabbix_get -s localhost -kmodbus_read["/dev/ttyS1 19200 N 8 1,22,32,4,f,1,0"]
0.989343

real	0m0.059s
user	0m0.000s
sys	0m0.020s

time zabbix_get -s localhost -kmodbus_read["/dev/ttyS1 19200 N 8 1,22,32,4,f,1,0"]
0.989396

real	0m0.078s
user	0m0.012s
sys	0m0.024s
But when the agent comes along, It can be more than a minute:

time zabbix_get -s localhost -kmodbus_read["/dev/ttyS1 19200 N 8 1,22,32,4,f,1,0"]
ZBX_NOTSUPPORTED: Connection timed out

real	0m10.064s
user	0m0.004s
sys	0m0.032s

time zabbix_get -s localhost -kmodbus_read["/dev/ttyS1 19200 N 8 1,22,32,4,f,1,0"]
0.989414

real	0m26.376s
user	0m0.012s
sys	0m0.024s

time zabbix_get -s localhost -kmodbus_read["/dev/ttyS1 19200 N 8 1,22,32,4,f,1,0"]
0.989414

real	0m26.376s
user	0m0.012s
sys	0m0.024s

time zabbix_get -s localhost -kmodbus_read["/dev/ttyS1 19200 N 8 1,22,32,4,f,1,0"]
0.989375

real	0m20.002s
user	0m0.008s
sys	0m0.028s
time zabbix_get -s localhost -kmodbus_read["/dev/ttyS1 19200 N 8 1,22,32,4,f,1,0"]

zabbix_get [29565]: Timeout while executing operation

real	**1m0.041s**
user	0m0.016s
sys	0m0.020s

v-zhuravlev, 2018-02-01:
> cannot find "zbx_module_history_write_cbs()" function in module "libzbxmodbus.so": /usr/local/lib/libzbxmodbus.so: undefined symbol: zbx_module_history_write_cbs

That doesn't sound right...

Netbulae, 2018-02-01:
It's an informational message it appears, it is since 3.2.4:

https://github.com/cavaliercoder/libzbxpgsql/issues/82

The log message is actually a debug-level message and is not an error condition. See src/libs/zbxmodules/modules.c in the Zabbix sources. Zabbix checks for all supported module functions and gracefully handles any optional functions by logging a message such as this.

Netbulae, 2018-02-01:
ZBX_HISTORY_WRITE_CBS	zbx_module_history_write_cbs(void);
https://www.zabbix.com/documentation/3.4/manual/config/items/loadablemodules

This function should return callback functions Zabbix server or proxy will use to export history of different data types. Callback functions are provided as fields of ZBX_HISTORY_WRITE_CBS structure, fields can be NULL if module is not interested in the history of certain type.

Netbulae, 2018-07-13:
Ah yes a bit of a weird sentence. I was testing manually and the agent gets the items every 10 seconds. When I disable the agent the manual zabbix_get runs fine, but when the agent runs I get timeouts.

I think we've traced it down to our systems rs-485 port not switching between receive and transmit properly. The usb rs-485 we originally used works much better as it has Auto Direction control.

I forgot to update the post and the project has been on hold for a while.

closed 2018-07-13 by Netbulae

sigtryggr, 2019-07-27:
Hi!
Sorry for deadposting, but the same problem had occured with RS485USB, Zabbix 4.2 and libzbxmodbus. When I trying to poll data from 485/USB Converter, no matter whether from zabbix server through agent or with zabbix_get, most of request returns "Connection timed out"; but sometimes, 2 of 10 requests, I'm getting the value from modbus device. Meanwhile, there're no such problems with modpoll, values are returning correctly. What can I do with this problem?