v-zhuravlev/libzbxmodbus #28: Negative numbers

Zmejs, 2018-05-14:
Your MODBUS module fails to interpret negative 64bit signed int correctly.
s - for MODBUS_SIGNED_INT
zabbix_get -s localhost -k "modbus_read[192.168.100.231,2,806,3,s,1,0]"
ZBX_NOTSUPPORTED: Illegal data value
S - for MODBUS_SIGNED_INT64
zabbix_get -s localhost -k "modbus_read[192.168.100.100,2,806,3,S,1,0]"
-4367981741133.000000
I - for MODBUS_INT64 (unsigned) 
zabbix_get -s localhost -k "modbus_read[192.168.100.100,2,806,3,I,1,0]"
18446739692842908600
l - for MODBUS_LONG, 32bit
zabbix_get -s localhost -k "modbus_read[192.168.100.100,2,806,3,l,1,0]"
18446744073709550517
Only last value is almost right, except one thing - it should be negative number -1099 in this case
Modpoll at same time returns correct values
 ./modpoll -m tcp 192.168.100.100 -a 2 -c 2 -t 4 -r 805
-- Polling slave... (Ctrl-C to stop)
[805]: -1
[806]: -1021

v-zhuravlev, 2018-05-17:
I think you need to check endianess and first reg used parameter.

Zmejs, 2018-05-17:
Tried everything.
Endianess unfortunately does nothing. Problem is that device allows me to read 2 32-bit words. If i try to read only 1, it returns error.
In either case first byte is FF, so its supposed to be negative anyway. But it isnt.
Examples with big and low endian
zabbix_get -s localhost -k "modbus_read[192.168.100.231,2,806,3,l,1,0]"
18446744073709551056
zabbix_get -s localhost -k "modbus_read[192.168.100.231,2,806,3,l,0,0]"
18446744073672589311
 zabbix_get -s localhost -k "modbus_read[192.168.100.231,2,806,3,S,1,0]"
-2422361554400.000000
zabbix_get -s localhost -k "modbus_read[192.168.100.231,2,806,3,S,0,0]"
154811241448538112.000000

Nothing even close to value, which im expecting.
For now only workaround for me is python script which does following
print (n & ((1 << 63) - 1)) - (n & (1 << 63))

v-zhuravlev, 2018-05-17:
why first_reg is 806 if in modpoll its 805?

Zmejs, 2018-05-17:
Well, its famous MODBUS numbering issue, same which is addressed with Your modules last parameter
0 - MODBUS_PDU_ADDRESSING is used. First data object has address 0. This is default value.

I can execute modpoll same way also
 ./modpoll -m tcp 192.168.100.231 -a 2 -c 2 -t 4 -r 806 -0
modpoll 3.4 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2013 proconX Pty Ltd
Visit http://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 2, start reference = 806 (PDU), count = 2
Communication.........: 192.168.100.231, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
[806]: -1
[807]: -648

v-zhuravlev, 2018-05-17:
Do you have device spec sheet?

v-zhuravlev, 2018-05-17:
What is the expected intege64 value? What is the value in hex? in two words 806-807.

v-zhuravlev, 2018-05-17:
64bit integers was implemented as experimental I never had a time to properly test it. https://github.com/v-zhuravlev/libzbxmodbus/pull/20/files

Zmejs, 2018-05-21:
Expected value is negative nuber. For example if i get 18446744073709551054, which is 0xFFFFFFFFFFFFFDCE hex, my expected value is -562

v-zhuravlev, 2018-07-10:
@i-ky , thanks! Can you please open same pull request here? https://github.com/v-zhuravlev/libmodbus This fork(because upstream libmodbus doesn't accept another [PR](https://github.com/stephane/libmodbus/pull/385)  is currently used in libzbxmodbus

i-ky, 2018-07-10:
Here you go: https://github.com/v-zhuravlev/libmodbus/pull/2

v-zhuravlev, 2018-07-10:
@i-ky , there is another small function in libzbxmodbus for 64bit float: https://github.com/v-zhuravlev/libzbxmodbus/blob/master/src/modbus.c#L148. Does it have the same problem?

i-ky, 2018-07-10:
I think no, it uses unsigned integers. Changing '+' to '|' will do no harm, but won't have any visible effect.

v-zhuravlev, 2018-07-10:
@Zmejs , can you please test again with new version of submodule libmodbus (available in master branch)

v-zhuravlev, 2018-08-16:
@Zmejs , I tested with the 64bit value you provided and works like you expect.
https://github.com/v-zhuravlev/libzbxmodbus/blob/tests/tests/test_modbus_64bit.py#L14

`modbus_read[{$MODBUS_HOST},{$MODBUS_SLAVEID},{$MODBUS_REG},3,S]`

v-zhuravlev, 2018-08-16:
Please reopen if still doesn't work for some reason

closed 2018-08-16 by v-zhuravlev

v-zhuravlev, 2018-09-01:
I also checked the spec you provided. It seem that 806 register is 2 words long, so 32bit INT not 64 bit. I implemented 32bit Integer in the latest libzbxmodbus (v0.8). This should work for your case.

PR #29 (2018-07-10): invalid Negative numbers
should fix #28, thanks to @i-ky