v-zhuravlev/libzbxmodbus #47: Create a private semaphore set.

i-ky, 2018-09-22:
In this piece of code `"."` is passed to `ftok()` function to create a key for the semaphore set:
https://github.com/v-zhuravlev/libzbxmodbus/blob/cf261eae6e6f63f886fa37fbeb663879d15bbc8b/src/modbus.c#L714-L721

A predetermined IPC resorce key is typically used so that several _standalone_ applications can access them for synchronization and data exchange. In our case semaphore set is used by forks of the same process and no external process is expected to take part in the operation. For this we can pass `IPC_PRIVATE` as `key` parameter to `semget()` call and it will always create a new semaphore set exclusively for us.

As a result:
* there is no need to handle the case when semaphore set with a given key has already been created;
* no risk of conflict between different instances of the module or with external applications using same key by accident.

I would also suggest moving semaphore-related code into a separate source file.

v-zhuravlev, 2018-09-23:
is it possible to create unique semaphores per connection_string?

i-ky, 2018-09-24:
It is possible to create its own semaphore for every `connection_string`, but there is a problem of sharing these semaphores across processes. Sharing is much easier at initialization stage, before Zabbix forks all its poller processes, but at this point we can't know how many unique `connection_string`s will be used and how many semaphores we need to create.

I think that current solution is good enough. I don't believe that hash collisions are such a big issue, especially after implementation of bulk requests, which should significantly reduce the number of requests needed to get all the data.

PR #53 (2019-10-03): refactored semaphore-related code
Should resolve #47:
* moved semaphore-related code into separate source file;
* changed semaphore set creation to use `IPC_PRIVATE`;
* improved error messages.
closed 2019-10-03 by v-zhuravlev