v-zhuravlev/libzbxmodbus #59: Value too small or too large

tonyskalski, 2019-04-18:
Since updating to Zabbix 4.0.6 I have been getting values too small or too large from a Campbell Scientific data logger. libzbxmodbus is working for a dozen other meters I am monitoring. There have been no changes to the data loggers for many years. I have been successfully retrieving data with libzbxmodbus for 18 months, prior to the upgrade.

Checking with 'modpoll -m tcp -t 3:float -a 1 -r 33 -c 1 -1 <ip_addr>' I get 57.0025 (the measurement is relative humidity). In Zabbix I am using 'modbus_read_registers[<ip_addr>, 1, 32, 4, float, LE]'  and I see 'Value 13364801568768.00000 is too small or too large'. 

I've taken packet captures and decoded the modbus messages for both modpoll and libzbxmodbus. I see the same data for each method (more or less - it varies by ~1% each reading).

I have tried all combinations of data type and endian-ness without luck. I have recompiled and reinstalled (via git) twice. What am I missing?

i-ky, 2019-04-18:
According to [this tool](https://www.h-schmidt.net/FloatConverter/IEEE754.html), binary representation of 13364801568768.00000 is `0x55427bc4`. And according to [another tool](https://www.scadacore.com/tools/programming-calculators/online-hex-converter/) we can get to a value close to what you expect (53.4418755, actually) by changing byte order from ABCD to BADC.

As far as I remember, there were issues in libmodbus regarding byte order (e.g. stephane/libmodbus/pull/441). This is one of the reasons libzbxmodbus has its own patched libmodbus. Unfortunately, it is indistinguishable from mainstream libmodbus (#22).

My best guess at this point is that during upgrade of Zabbix you accidentally upgraded libmodbus as well and mainstream version has overwritten the patched one.

tonyskalski, 2019-04-18:
I had the same thought (an unpatched libmodbus). So before the last re-install of libzbxmodbus, I removed the libmodbus files. For the reinstall I did `git clone --recursive https://github.com/v-zhuravlev/libzbxmodbus.git` (and autogen and config and make). This should get the patched version of libmodbus right?

i-ky, 2019-04-18:
Yes, it should. I would check the output of
```
$ ldd libzbxmodbus.so
```
and verify the modification time of `libmodbus.so` it points to. I hope you have restarted Zabbix after recompiling libs.

tonyskalski, 2019-04-18:
While I would swear on stack of holy texts that I tried "0" for the endian-ness, I just re-tried and it is working. I did discover that I need to specify "float" and not "f" for the data type. So I will switch all my keys to use "float, MLE" so that it is more explicit. Thanks for the help!

closed 2019-04-18 by tonyskalski

i-ky, 2019-04-18:
I believe with your test command `modpoll` will assume little endian byte order. The fact that you need `MLE` in Zabbix may be a hint that you have unpatched libmodbus.

tonyskalski, 2019-04-18:
It certainly would seem that way, but:
```
[root@mon lib]# ldd libzbxmodbus.so
	linux-vdso.so.1 =>  (0x00007ffeb2ff2000)
	libmodbus.so.5 => /etc/zabbix/lib/libmodbus.so.5 (0x00007f076a9fe000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f076a631000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f076ae10000)
[root@mon lib]# ls -l /etc/zabbix/lib/libmodbus.so.5
lrwxrwxrwx 1 root root 18 Apr 18 12:06 /etc/zabbix/lib/libmodbus.so.5 -> libmodbus.so.5.1.0
```
which is what I compiled via git around lunchtime. I am still confused, but glad it is working.