v-zhuravlev/libzbxmodbus #60: Question

javadgo, 2019-05-08:
Hi 
Thank you very much for this great repository. I have question and I would be very glad if you can give me an answer.
Can I load libzbxmodbus module both in **Zabbix-server** and **Zabbix-proxy** in **one computer** to pull data from a serial port (ttyUSB0)? Is there any conflict or data loss in this structure?
Info: Zabbix-proxy in my structure will send data for a remote master Zabbix-server.

i-ky, 2019-05-08:
Until #53 is merged into master, you can make all module instances (loaded by server(s), proxy(-ies) and agent(s)) running on the same machine use a common semaphore set which should allow them to synchronize access to the same serial port. However, when one of Zabbix daemons stops, module will remove this semaphore set and this will cause other module instances to fail. So just to be safe and future-proof, I'd suggest to install a single Zabbix agent, which will be the only one who accesses that serial port.

javadgo, 2019-05-11:
I have implemented the suggested structure. It work and now I receive data in zabbix servers local and  zabbix servers master.
The problem is, I have some data loss which I don't have before (when I load libzbxmodbus module both in Zabbix-server and Zabbix-proxy).
Is there any downside in loading libzbxmodbus in zabbix-agent or there is some misconfiguration.

Data loss can be seen in this picture:
[image]

i-ky, 2019-05-11:
> Is there any downside in loading libzbxmodbus in zabbix-agent

No, nothing I'm aware of. My best guess is that you are overloading agent with lots of checks. You can try to increase the number of agent's listener processes (`StartAgents`, if I recall correctly) if you are using passive checks. But it is difficult to diagnose the problem without knowing what's happening in Zabbix log files.

javadgo, 2019-05-15:
Your thought was true and StartAgents solved the problem.
Now I have another question! 😀
I have a 4-channel relay board which support modbus **read** and **write** function.
I can get 4-channel relay values in zabbix thanks to libzbxmodbus module, the question here is: Is there any way for sending write command to the registers? (for example using zabbix-agent and zabbix-send)
I see git readme file and there isn't any function for write ==>
- modbus_function:
     Modbus function in integer form:
     1 - for **READ** COILS
     2 - for **READ** DISCRETE INPUT STATUS
     3 - for **READ** HOLDING REGISTERS
     4 - for **READ** INPUT REGISTERS

i-ky, 2019-05-15:
> Is there any way for sending write command to the registers?

Using this module, no.

closed 2019-07-06 by v-zhuravlev