v-zhuravlev/libzbxmodbus #62: Timeouts when requesting over RS485/USB

sigtryggr, 2019-07-29:
Hi!
The problem had occured with RS485USB, Zabbix 4.2 and libzbxmodbus. When I trying to poll data from 485/USB Converter, no matter whether from zabbix server through agent or with zabbix_get, most of request returns "Connection timed out"; but sometimes, 2 of 10 requests, I'm getting the value from modbus device. Meanwhile, there're no such problems with modpoll, values are returning correctly. What can I do with this problem?

TelDragon, 2020-05-26:
I also encountered the same problem

i-ky, 2020-05-26:
Could you please show an example of item key and modpoll command?

TelDragon, 2020-05-29:
> I also encountered the same problem

I have solved it now, sorry.
It's my own setting parameters.

i-ky, 2020-05-29:
Then it is a question to @sigtryggr.

i-ky, 2021-05-03:
Almost a year without response. I'm going to close the issue. Feel free to reopen if the issue still persists.

closed 2021-05-03 by i-ky