v-zhuravlev/libzbxmodbus #63: 2 protocol

bobbyps23, 2019-08-13:
Hai, sorry for another question, I've success on installing libmodbuszbx. Here's the problem, can it installed along with SNMP?if it can, then should I change the prefix to /var/lib/module?because I need to monitoring this 2 protocol. Thanks

i-ky, 2019-08-13:
I don't know what you actually mean by "installing SNMP". Zabbix agent protocol and SNMP are two completely independent protocols and their configurations don't affect each other.

bobbyps23, 2019-08-14:
I'm sorry, I forgot that SNMP is already installed from the start. But it LoadModulePath it's in the zabbix_server.conf too, so I can't put LoadModulePath for libzbxmodbus there, it will make zabbix server stopped because it will search for module.so for modbus in SNMP's module Path directory. So what should I do so I can read SNMP and Modbus in the same time without causing error in zabbix server?

i-ky, 2019-08-14:
Zabbix server supports SNMP out of the box, you don't need any modules for that...

In case you need to use two or more loadable modules, they all have to be located in the same directory (`LoadModulePath`). Latest versions of Zabbix also support absolute path in `LoadModule` (see [the feature request](https://support.zabbix.com/browse/ZBXNEXT-5179) for more details).

bobbyps23, 2019-08-14:
So, if I've already put the module for modbus in the same directory(LoadModulePath) as the SNMP's directory, do I also need to write libmodbus.so in LoadModule too or not? I'm so sorry for asking too many, because I'm still vague about it.

i-ky, 2019-08-14:
Yes, one `LoadModule` for every module you want to use.

bobbyps23, 2019-08-14:
Okay then. Great thanks a lot.

v-zhuravlev, 2019-08-14:
@i-ky , thanks for keeping this repo responsive :)

closed 2019-08-14 by v-zhuravlev