v-zhuravlev/libzbxmodbus #64: Reuse existing connection.

radjah, 2019-12-16:
Пытаюсь писать данные с эмулятора slave modbus по TCP.
Поставил чтение регистра раз в секунду. При каждом чтении создаётся новое соединение, что в итоге переполняет количество доступных соединение эмулятора, и в место получения данных возникает таймаут.
Если в эмуляторе прописать пул в 100 соединений, то до достижения лимита первое успевает сброситься.
В key прописал следующее:
`modbus_read_registers[tcp://192.168.1.3,1,2,3,s,0]`
В конфиге выставил Timeout=1

Версия zabbix 4.4.3

v-zhuravlev, 2019-12-16:
спасибо за интерес к модулю. К сожалению, я сейчас с ним не работаю, и пока нет времени разобраться, что к чему. Но буду рад Pull Request, который сделают данную оптимизацию работы модуля.

i-ky, 2019-12-18:
> Поставил чтение регистра раз в секунду.

Насколько это реалистичный сценарий? Действительно ли требуется считывать данные раз в секунду?

radjah, 2019-12-19:
> Насколько это реалистичный сценарий? Действительно ли требуется считывать данные раз в секунду?

Испытания техники на работе провожу. Я бы еще чаще опрос поставил, если бы Zabbix это позволял.

Так-то испытательный комплекс у нас до 10 раз в секунду может опрашивать контроллеры по modbus tcp.

i-ky, 2019-12-21:
> до 10 раз в секунду

Для Zabbix даже опрос раз в секунду - это что-то неслыханное. Если помножить опрос одной метрики 10 раз в секунду на, скажем, 10 метрик на устройство и, скажем, 1000 устройств, то получатся внушительные 100 тысяч значений в секунду. Редкая инсталляция Zabbix такое потянет.

Я это к тому, что даже если модуль сможет опрашивать регистры с такой частотой, не факт, что остальная инфраструктура Zabbix справится с таким потоком данных. Реализация пула соединений на стороне модуля - это довольно сложная задача, дающая незначительный, на мой взгляд, выигрыш. Если нужно опрашивать много метрик с одного устройства, то [bulk запросы](https://github.com/v-zhuravlev/libzbxmodbus#5-bulk-data-collection) должны помочь с количеством соединений.

radjah, 2019-12-22:
@i-ky, не в этом дело. Можно читать хоть один регистр, хоть 10 с помощью __одного__ запроса, всё равно каждый запрос порождает новое соединение, а старое весить в ожидание таймаута завершения.
Проверил на днях на [живой железке](https://www.fastwel.ru/products/fastwel-io/programmiruemye-kontrollery-uzla-seti/programmiruemyy-kontroller-uzla-seti-ethernet-protokol-peredachi-modbustcp-10-100-mbit-s-100m-vedushch/). Она всё это дела всё же переварила и не подавилась.
До этого пробовал [OpenMUC](https://www.openmuc.org/) и [приблуду](https://javalibs.com/artifact/org.openmuc.framework/openmuc-driver-modbus) для modbus. Оно делает одно соединение до устройства и работает внутри него, но совсем не умеет в порядок байт для float, что в моём случае критично.
Короче, два стула.

i-ky, 2019-12-22:
> каждый запрос порождает новое соединение

Я согласен, что это не оптимально, но так работает сам Zabbix, и такая же логика работы навязывается модулям. В двух из трёх возможных сценариев использования модулей (_Simple check_ и _Zabbix agent_) код модуля может вызываться разными процессами Zabbix'а (`poller` или `listener`, соответственно). Если у каждого процесса будет свой пул открытых соединений, то это может привести к излишнему потреблению ресурсов. Реализовать общий пул соединений или специальный процесс/демон, который будет заниматься непосредственно опросом устройств, будет слишком сложно, если вообще возможно.

В третьем сценарии (_Zabbix agent (active)_) код модуля вызывается одним процессом Zabbix'а (процессом `active`), так что оставление соединения открытым для повторного использования имело бы смысл. Но проблема в том, что у модуля нет никакой возможности (по крайней мере в официальном API) различить эти три сценария.

Так что у разработчиков модулей тоже "два стула"...

> а старое весить в ожидание таймаута завершения

Это может быть багом в модуле или библиотеке libmodbus. Расскажите подробнее о том, как Вы используете модуль, и как Вы диагностировали такую проблему.

radjah, 2019-12-22:
> как Вы используете модуль
Загрузил модуль в агент, добавил item типа "Zabbix agent" с запросом пачки регистров и куча зависимых от него item-ов для разбора полученного json.
В качестве эмулятора использовал вот это http://www.plcsimulator.org/
В левом верхнем углу окна есть счётчик соединений, по нему и ориентировался.

i-ky, 2019-12-23:
А возможно это ~баг~ особенность самого [эмулятора](http://www.plcsimulator.org/documentation/faq):
> **Q:** Why does the simulator not allow me to connect over TCP after 10 retries?
**A:** This was a simulation option emulating non-graceful socket close (no FIN-ACK) to simulate broken WANs, if I can get funding I may make this a settable option.

radjah, 2020-01-14:
Испытал в боевых условиях на Raspberry Pi 3.
С помощью модуля вычитывал где-то 30 регистров в JSON раз в секунду и логировал с помощью зависимых item-ов.
Контроллер по ссылке из комментария выше с ума не сходил, данные писались чётко.
Видимо зря поднял панику.

closed 2021-05-03 by i-ky