v-zhuravlev/libzbxmodbus #68: multiple serial port possibility

javadgo, 2020-04-16:
Hi
Thanks a bunch for this great repo, Vitaly you are the best.
I have a question and I would be very grateful if anyone can give me an answer.
As it's well described in  README file, all serial port configuration is in items key definition in zabbix UI.
My PC have two serial port, can I connect two different modbus line to each serial ports like below structure?

[image]

Main question is, is it possible to for libzbxmodbus to handle multiple serial port at the same time?

i-ky, 2020-04-16:
I don't see anything related among [limitations](https://github.com/v-zhuravlev/libzbxmodbus#7-known-issues-and-limitations), so it should be possible. I think you should try!

i-ky, 2021-05-03:
I am going to assume that everything went fine and close the issue. @javadgo, feel free to reopen or create new issues if you have more questions or experience any problems.

closed 2021-05-03 by i-ky