v-zhuravlev/libzbxmodbus #70: libzbxmodbus

behzad831371, 2020-05-18:
Hi,
I tested my connection with modpoll and received the data. When I try to read the modbus in zabbix, it seems that the connection is established and no unsupported item, but the data is different than what it should be and is always 0.
Thanks.

[image]
[image]
[image]

v-zhuravlev, 2020-05-18:
Check things like endianess and so on and whether they match in modpoll and zabbix modbus. Those things can be set by using optional params in zabbix item key. To speed up troubleshooting - I recommend to test modbus using `zabbix_get` or `zabbix_agentd -t` cli commands.

behzad831371, 2021-01-21:
Thank you for your answer
But I am a very beginner can I get more help from you?
I have been involved for some time, I tried but unfortunately it did not work

closed 2021-01-21 by behzad831371