v-zhuravlev/libzbxmodbus #71: Zabbix version 5.0

OG0505, 2020-06-26:
i am trying to do the following command:
./configure --prefix=/etc/zabbix --enable-zabbix-5.0

And i am getting the following error code:
Please specify the version of Zabbix module will be used with: by providing either
--enable-zabbix-2 or --enable-zabbix-3 or --enable-zabbix-3.2 key but no more than one.

What am i doing wrong?
My zabbix version is 5.0.1

I tried the following commands also and i am getting the same error:
./configure --prefix=/etc/zabbix --enable-zabbix-5
./configure --prefix=/etc/zabbix --enable-zabbix-5.0.1

[image]

i-ky, 2020-06-26:
> What am i doing wrong?

Nothing. This part hasn't been updated since Zabbix 3.2 and no surprise that nowadays it is a bit confusing.

Luckily, Zabbix headers didn't change much either (if at all), so I guess you will be fine if you use `--enable-zabbix-3.2` with Zabbix 5.0.

OG0505, 2020-06-26:
Thankyou it worked! with the 3.2 key

closed 2020-06-26 by OG0505