v-zhuravlev/libzbxmodbus #72: Check SlaveID parameter

OG0505, 2020-07-02:
Hi,

I configured the modbus items on the Raspberry Pi just like the manual says but i am getting a error on the modbus request item.
- Check slaveid parameter.

I dont know what to do know. 
Did i forget something?
How do i know if the modbus is working right know?

[image]
[image]
[image]
[image]
[image]

i-ky, 2020-07-02:
Hi!

It looks like you are using [item testing](https://www.zabbix.com/documentation/current/manual/config/items/item#testing) functionality. Unfortunately, I don't have a hands on experience with this feature. And unfortunately, Zabbix documentation does not say anything about _Macros_. But I suppose Zabbix has detected that item key you want to test depends on macros and you are supposed to provide their values in input boxes next to &rArr;. Alternatively, for the sake of testing, you can replace macros with their values in the item key itself.

i-ky, 2021-05-03:
Almost a year without response. I am going to close the issue. @OG0505, feel free to reopen if the issue still persists.

closed 2021-05-03 by i-ky