v-zhuravlev/libzbxmodbus #81: unsupported module "libzbxmodbus.so" version: 1

liuhuan94, 2021-11-17:
Hello developer, I installed and deployed libzbxmodbus in Zabbix5.4, but it is not supported, please help me, thank you!
[image]

i-ky, 2021-11-17:
@liuhuan94, did you compile libzbxmodbus yourself? If yes, what was your `./configure` command? For Zabbix 3.2 and later something like this should work:
```
./configure --enable-zabbix-3.2 ...
```

liuhuan94, 2021-11-17:
> @liuhuan94, did you compile libzbxmodbus yourself? If yes, what was your `./configure` command? For Zabbix 3.2 and later something like this should work:
> 
> ```
> ./configure --enable-zabbix-3.2 ...
> ```
Of course, zabbix log still shows failure after compilation
[image]

liuhuan94, 2021-11-17:
[image]
[image]

liuhuan94, 2021-11-17:
[image]

i-ky, 2021-11-17:
Just to double check. Did you run `make` after `./configure --enable-zabbix-3.2`?

liuhuan94, 2021-11-17:
> Just to double check. Did you run `make` after `./configure --enable-zabbix-3.2`?
Ah yes，I installed it through Git. Now I uninstall the previous installation and download the installation package again for compilation. No error has been reported so far.Thank you！

i-ky, 2021-11-17:
You are welcome!

closed 2021-11-17 by i-ky