v-zhuravlev/libzbxmodbus #82: modbus.read timeout

liuhuan94, 2021-11-19:
Hi developer, I don't understand why modbus.read shows timeout when testing port pass.I used nc -zv test port connect show sucess

[image]
[image]

i-ky, 2021-11-19:
@liuhuan94, this error message comes from libmodbus, which is used by this module to perform actual communications. If I were you, I would try to capture traffic while doing `modbus.read` (using `tcpdump` & Co.), this may shed some light on the issue.

liuhuan94, 2021-11-19:
I looked at it and it worked。
[image]

i-ky, 2021-11-19:
@liuhuan94, is the issue resolved?

liuhuan94, 2021-11-22:
> @liuhuan94, is the issue resolved?

used the modbus_read_registers [TCP: / / 10.148.20.24:501,1,2,3, s, 0] can get data,But if you poll all the data in the register, the connection is refused

[image]

liuhuan94, 2021-11-22:
Sorry，The port number above is wrong. It should be this one
[image]

i-ky, 2021-11-22:
@liuhuan94, like I said [above](https://github.com/v-zhuravlev/libzbxmodbus/issues/82#issuecomment-973911372), I have no way of helping you unless you capture the traffic.

liuhuan94, 2021-11-22:
> @liuhuan94，就像我[上面](https://github.com/v-zhuravlev/libzbxmodbus/issues/82#issuecomment-973911372)说的，除非你抓住流量，否则我无法帮助你。

display success

[image]

i-ky, 2021-11-22:
@liuhuan94, have you tried testing your item key with `zabbix_get` or `zabbix_agentd -t`?

liuhuan94, 2021-11-23:
> @liuhuan94, have you tried testing your item key with `zabbix_get` or `zabbix_agentd -t`?

Zabbix_get requires the installation of Zabbix Agent in the data acquisition terminal, but my device is an IOT temperature and humidity collector, so I cannot use ZabbiX_GET for testing

i-ky, 2021-11-23:
@liuhuan94, you can install `zabbix_get` and Zabbix agent into the same box where your Zabbix server is. And query your IOT device using agent just like you do with Zabbix server, but using agent will help isolate the issue, hopefully.

liuhuan94, 2021-11-25:
>

liuhuan94, 2021-11-25:
> @liuhuan94, you can install `zabbix_get` and Zabbix agent into the same box where your Zabbix server is. And query your IOT device using agent just like you do with Zabbix server, but using agent will help isolate the issue, hopefully.

The problem has been solved, I found the problem, the parameter setting and register reading position is wrong, causing the connection timeout.Thank you for guiding me through mistakes.

closed 2021-11-25 by liuhuan94