v-zhuravlev/libzbxmodbus #84: undefined symbol: modbus_new_rtutcp

gbs098, 2023-01-16:
61681:20230116:130127.924 Zabbix Agent stopped. Zabbix 6.0.12 (revision 126aa2f53e9).
 61732:20230116:130140.167 Starting Zabbix Agent [Zabbix server]. Zabbix 6.0.12 (revision 126aa2f53e9).
 61732:20230116:130140.168 using configuration file: /etc/zabbix/zabbix_agentd.conf
 61732:20230116:130140.168 cannot load module "libzbxmodbus.so": /etc/zabbix/lib/libzbxmodbus.so: undefined symbol: modbus_new_rtutcp

i-ky, 2023-01-16:
@gbs098, could you please provide more information? How did you installed the module? How long ago? Did you make any changes recently that could have caused the issue?

My guess at this point is that it is related to #22. Maybe a system-wide libmodbus has been updated recently and overwritten patched libmodbus provided with the module.

gbs098, 2023-01-16:
Hello,
Today I tried both methods
tar zxvpf libzbxmodbus-0.9.tar.gz
cd libzbxmodbus-0.9
./configure --prefix=/etc/zabbix --enable-zabbix-[2|3|3.2]
make
make install
--------------------------
And this one:
https://github.com/v-zhuravlev/libzbxmodbus/wiki/Install-from-Github-repo

i-ky, 2023-01-16:
> ./configure --prefix=/etc/zabbix --enable-zabbix-[2|3|3.2]

You need to choose one of `2`, `3`, `3.2` (for Zabbix 6.0 it should be `3.2`).

Could you please show the output of `ldd /etc/zabbix/lib/libzbxmodbus.so`?

gbs098, 2023-01-16:
linux-vdso.so.1 (0x00007ffd771e3000)
        libmodbus.so.5 => /etc/zabbix/lib/libmodbus.so.5 (0x00007f659060c000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f659042e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f6590623000)

gbs098, 2023-01-16:
I used 3.2

gbs098, 2023-01-16:
Done,
Thx
I made 'apt-get --purge remove libmodbus-dev'
And reinstall zabbix-agent now it works

closed 2023-01-16 by gbs098