v-zhuravlev/libzbxmodbus #85: Failed to build libzbxmodbus-0.9 for Zabbix 7.0

snowuyl, 2024-11-15:
wget https://github.com/v-zhuravlev/libzbxmodbus/releases/download/v0.9/libzbxmodbus-0.9.tar.gz
tar xvf libzbxmodbus-0.9.tar.gz
cd libzbxmodbus-0.9
./configure --prefix=/etc/zabbix --enable-zabbix-7.0
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
configure: error: 
------------------------------------------
Please specify the version of Zabbix module will be used with: by providing either
--enable-zabbix-2 or --enable-zabbix-3 or --enable-zabbix-3.2 key but no more than one.
------------------------------------------

i-ky, 2024-11-15:
@snowuyl, try `--enable-zabbix-3.2` instead of `--enable-zabbix-7.0`.

snowuyl, 2024-11-15:
@i-ky Thanks for your great support! libzbxmodbus-0.9 can be built successfully after changing --enable-zabbix-7.0 to --enable-zabbix-3.2.

closed 2024-11-15 by snowuyl